esphome/firmware #288: dfd

bmsroque, 2024-12-21:
dfgdf

monotonus, 2025-01-08:
Exactly the same behaviour here. Got my M5 Atom Echo yesterday, flashed it via web, added it to Home Assistant, but get no voice.

[image]

monotonus, 2025-02-04:
Seems to not be only a Atom Echo problem. I have this also with the assist in general. I'm running Home Assistant Container. @bmsroque Are you running a full HAOS or also a docker container?

bmsroque, 2025-02-04:
> Seems to not be only a Atom Echo problem. I have this also with the assist in general. I'm running Home Assistant Container. [@bmsroque](https://github.com/bmsroque) Are you running a full HAOS or also a docker container?
> 
> Seems I have fixed it for me. I have no "default_config:" in my configuration.yaml, so I needed to also add the assist_pipeline: to the config (I already had conversation:) added. This didn't help and then I found this post where it was mentioned that also media_source: is necessary for TTS. So added this also and now it works.
> 
> [home-assistant/home-assistant.io#27271 (comment)](https://github.com/home-assistant/home-assistant.io/issues/27271#issuecomment-1537798255)
> 
> https://www.home-assistant.io/integrations/default_config/
> 
> I hope this helps you.

I´m using HA in VM with proxmox. Can you tell me what code did you used in Echo yaml file please. I will try this too. Thank you.

jesserockz, 2025-02-10:
tts-failed has nothing to do with the firmware or even the device. It is entirely a failure in Home Assistant or in the HA Cloud tts engine. Please open an issue there.

closed 2025-02-10 by jesserockz