esphome/firmware #292: Missing repository for esphome-web-f47e8c.yaml

huseman21, 2025-01-15:
Repository seems to be gone for the esphome-web-f47e8c.yaml

 source: github://RASPIAUDIO/esphomeLuxe@main

Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/raspiaudio-muse-proto/.pioenvs/raspiaudio-muse-proto/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

AadeeshJainB2K, 2025-02-08:
same here bro any solutions

AadeeshJainB2K, 2025-02-08:
bro i found the solution it would be the problem with ethernet cable of your home assistant

jesserockz, 2025-02-10:
Seems like a temporary internet failure. Please reopen if the issue persists

closed 2025-02-10 by jesserockz