esphome/firmware #293: Install m5stack-atom-echo.yaml

20flint12, 2025-01-17:
Failed to resolve component 'jsmn'.

20flint12, 2025-01-17:
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
packages:
  # m5stack.atom-echo: github://esphome/firmware/media-player/m5stack-atom-echo.yaml@main
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.adopted.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
api:
  encryption:
    key: !secret athom_echo_encription_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ericvb, 2025-01-21:
Please see : https://github.com/esphome/firmware/issues/286

jesserockz, 2025-02-10:
Duplicate of #286

closed 2025-02-10 by jesserockz