esphome/firmware #297: 2025.2.0 breaks audio builds for M5Stack Atom Echo

michaelkleinhenz, 2025-02-20:
2025.2.0 builds for the M5Stack Atom Echo are breaking on compiling the audio integration. Unchanged project worked on 2024.12.4. 

Project uses the following version config:

```
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  min_version: 2024.9.0

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```

(removing the IDF version force breaks compilation as well, this setup was working on 2024.12.4)

Log:

```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/m5stack-atom-echo-fdedc0.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Updating https://github.com/jesserockz/esphome-components.git@None
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING RMT_LED_STRIP support for IDF version < 5 is deprecated and will be removed soon.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Updating https://github.com/espressif/esp-nn@v1.1.0
INFO Compiling app...
Processing m5stack-atom-echo-fdedc0 (board: m5stack-atom; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40408.0 (4.4.8) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- noise-c @ 0.1.6
|-- ESPMicroSpeechFeatures @ 1.1.0
|-- esp-audio-libs @ 1.1.1
Generating assembly for .pioenvs/m5stack-atom-echo-fdedc0/duer_profile.S
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/api_connection.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/api_server.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/list_entities.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/proto.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/api/user_services.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/audio/audio.o
Compiling .pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/audio/audio_decoder.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
src/esphome/core/helpers.h:85:59: note:   template argument deduction/substitution failed:
src/esphome/components/audio/audio_decoder.cpp:79:81: error: template argument 1 is invalid
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
src/esphome/components/audio/audio_decoder.cpp:80:25: error: base operand of '->' is not a pointer
       this->wav_decoder_->reset();
                         ^~
src/esphome/components/audio/audio_decoder.cpp: In member function 'esphome::audio::FileDecoderState esphome::audio::AudioDecoder::decode_wav_()':
src/esphome/components/audio/audio_decoder.cpp:314:5: error: 'esp_audio_libs' has not been declared
     esp_audio_libs::wav_decoder::WAVDecoderResult result = this->wav_decoder_->decode_header(
     ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:9: error: 'result' was not declared in this scope
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/m5stack-atom-echo-fdedc0/src/esphome/components/audio/audio_decoder.o] Error 1
========================= [FAILED] Took 120.10 seconds =========================
```

michaelkleinhenz, 2025-02-20:
Cleaning the build files does not change the behaviour.

michaelkleinhenz, 2025-02-20:
Removing the forced `esp-idf` version results in a very weird error:

```
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Current idf-component-manager default lock file version is 2.0.0, but found 1.0.0 in /data/build/m5stack-atom-echo-fdedc0/dependencies.lock. Recreating lock file with the current version.
Manifest files have changed, solving dependencies.
.....NOTICE: Updating lock file at /data/build/m5stack-atom-echo-fdedc0/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.1.5)
-- Configuring incomplete, errors occurred!
See also "/data/build/m5stack-atom-echo-fdedc0/.pioenvs/m5stack-atom-echo-fdedc0/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'jsmn'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:308 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)
```

michaelkleinhenz, 2025-02-20:
This might be related to this: https://github.com/esphome/wake-word-voice-assistants/issues/42

mbay0r, 2025-02-20:
```
esphome:
  compile_process_limit: 10
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.9.0
  name_add_mac_suffix: False
  platformio_options:
    board_build.flash_mode: dio
  on_boot:
    priority: 600
    then:
      - lambda: |-
          auto volume_component = id(speaker_volume);
          volume_component->set_volume(0.9);
      - script.execute: draw_display
      - delay: 30s
      - if:
          condition:
            lambda: return id(init_in_progress);
          then:
            - lambda: id(init_in_progress) = false;
            - script.execute: draw_display

esp32:
  board: esp32s3box
  flash_size: 16MB
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_AUDIO_BOARD_CUSTOM: "y"
      CONFIG_ESP32_S3_BOX_3_BOARD: "y"
    components:
      - name: esp32_s3_box_3_board
        source: github://jesserockz/esp32-s3-box-3-board@main
        refresh: 0s
```

```
Compiling .pioenvs/esp32-s3-box-3/src/esphome/components/display/display_buffer.o
In file included from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/components/audio/audio_decoder.h:103:19: error: 'esp_audio_libs' was not declared in this scope
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 1 is invalid
   std::unique_ptr<esp_audio_libs::wav_decoder::WAVDecoder> wav_decoder_;
                                                          ^
src/esphome/components/audio/audio_decoder.h:103:58: error: template argument 2 is invalid
src/esphome/components/audio/audio_decoder.cpp: In member function 'esp_err_t esphome::audio::AudioDecoder::start(esphome::audio::AudioFileType)':
src/esphome/components/audio/audio_decoder.cpp:79:40: error: 'esp_audio_libs' was not declared in this scope
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                        ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:28: error: parse error in template argument list
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:79:81: error: no matching function for call to 'make_unique<<expression error> >()'
       this->wav_decoder_ = make_unique<esp_audio_libs::wav_decoder::WAVDecoder>();
                                                                                 ^
In file included from src/esphome/components/speaker/speaker.h:12,
                 from src/esphome/components/audio/audio_transfer_buffer.h:8,
                 from src/esphome/components/audio/audio_decoder.h:6,
                 from src/esphome/components/audio/audio_decoder.cpp:1:
src/esphome/core/helpers.h:85:59: note: candidate: 'template<class T, class ... Args> std::unique_ptr<T> esphome::make_unique(Args&& ...)'
 template<typename T, typename... Args> std::unique_ptr<T> make_unique(Args &&...args) {
                                                           ^~~~~~~~~~~
[... 13 lines cut ...]
         ^~~~~~
src/esphome/components/audio/audio_decoder.cpp:317:19: error: 'esp_audio_libs' has not been declared
     if (result == esp_audio_libs::wav_decoder::WAV_DECODER_SUCCESS_IN_DATA) {
                   ^~~~~~~~~~~~~~
src/esphome/components/audio/audio_decoder.cpp:318:78: error: base operand of '->' is not a pointer
       this->input_transfer_buffer_->decrease_buffer_length(this->wav_decoder_->bytes_processed());
                                                                              ^~
src/esphome/components/audio/audio_decoder.cpp:321:29: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                             ^~
src/esphome/components/audio/audio_decoder.cpp:321:68: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                    ^~
src/esphome/components/audio/audio_decoder.cpp:321:104: error: base operand of '->' is not a pointer
           this->wav_decoder_->bits_per_sample(), this->wav_decoder_->num_channels(), this->wav_decoder_->sample_rate());
                                                                                                        ^~
src/esphome/components/audio/audio_decoder.cpp:323:49: error: base operand of '->' is not a pointer
       this->wav_bytes_left_ = this->wav_decoder_->chunk_bytes_left();
                                                 ^~
src/esphome/components/audio/audio_decoder.cpp:326:26: error: 'esp_audio_libs' has not been declared
     } else if (result == esp_audio_libs::wav_decoder::WAV_DECODER_WARNING_INCOMPLETE_DATA) {
                          ^~~~~~~~~~~~~~
*** [.pioenvs/esp32-s3-box-3/src/esphome/components/audio/audio_decoder.o] Error 1
========================== [FAILED] Took 4.98 seconds ==========================
```

trevm8, 2025-02-24:
I have the same problem

michaelkleinhenz, 2025-02-25:
This resolved now. See https://github.com/esphome/wake-word-voice-assistants/issues/42#issuecomment-2678350895. This is now using a different audio setup that is also working on 2025.2.0.

closed 2025-02-25 by michaelkleinhenz

darinlarimore, 2025-02-25:
Hey thanks @michaelkleinhenz for documenting this issue. 

I resolved this by using the new yaml file here:
https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml

hitnrun30, 2025-02-27:
I am not understanding, I use the https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.adopted.yaml which is used through Home Assistant with the ESPHome Add on.

viktordd, 2025-02-27:
I'm still having an issue with no audio playing on my M5Stack Atom Echo. I'm using the [https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml).
I tried setting the refresh to a day so I make sure I have latest, cleaning build files and reinstalling, but still have the same issue.

Here's my Yaml
``` yaml
substitutions:
  name: m5stack-atom-echo
  friendly_name: M5Stack Atom Echo
  micro_wake_word_model: okay_nabu # okay_nabu, hey_jarvis and hey_mycroft are supported

packages:
  m5stack.atom-echo-voice-assistant:
    url: https://github.com/esphome/wake-word-voice-assistants
    files: [m5stack-atom-echo/m5stack-atom-echo.yaml]
    ref: main
    refresh: 1d

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: !secret m5stack_encryption_key

ota:
  - id: !extend ota_esphome
    password: !secret m5stack-atom-echo-88878c-ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  on_connect:
    - delay: 5s
```

Here's log output from me trying to play media (just using piper to say something)
```
[12:46:42][D][media_player:073]: 'M5Stack Atom Echo 88878c' - Setting
[12:46:42][D][media_player:080]:   Media URL: http://{local_ip}/api/esphome/ffmpeg_proxy/cb46e1fcd1ebaf169af02d29a19fe897/RtyAXHJ6e7cMuNiomlNi0w.wav
[12:46:42][D][media_player:086]:  Announcement: yes
[12:46:42][D][micro_wake_word:195]: State changed from DETECTING_WAKE_WORD to STOP_MICROPHONE
[12:46:42][D][light:036]: 'M5Stack Atom Echo 88878c' Setting:
[12:46:42][D][light:051]:   Brightness: 60%
[12:46:42][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[12:46:42][D][light:036]: 'M5Stack Atom Echo 88878c' Setting:
[12:46:42][D][light:051]:   Brightness: 100%
[12:46:42][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[12:46:42][D][speaker_media_player:420]: State changed to ANNOUNCING
[12:46:42][W][component:237]: Component speaker.media_player took a long time for an operation (51 ms).
[12:46:42][W][component:238]: Components should block for at most 30 ms.
[12:46:42][D][micro_wake_word:129]: Stopping Microphone
[12:46:42][D][micro_wake_word:195]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[12:46:42][D][esp-idf:000]: I (570720) I2S: DMA queue destroyed

[12:46:42][D][speaker_media_player.pipeline:114]: Reading WAV file type
[12:46:42][D][micro_wake_word:195]: State changed from STOPPING_MICROPHONE to IDLE
[12:46:42][D][ring_buffer:034][speaker_task]: Created ring buffer with size 1920
[12:46:42][D][esp-idf:000][speaker_task]: I (570761) I2S: queue free spaces: 5

[12:46:42][D][esp-idf:000][speaker_task]: I (570767) I2S: DMA Malloc info, datalen=blocksize=960, dma_buf_count=4

[12:46:42][D][i2s_audio.speaker:116]: Started Speaker
[12:46:46][D][speaker_media_player:420]: State changed to IDLE
[12:46:47][D][esp-idf:000][speaker_task]: I (575178) I2S: DMA queue destroyed

[12:46:47][D][i2s_audio.speaker:129]: Stopped Speaker
[12:46:47][W][micro_wake_word:183]: Wake word is already stopped
[12:46:48][D][light:036]: 'M5Stack Atom Echo 88878c' Setting:
[12:46:48][D][light:051]:   Brightness: 60%
[12:46:48][D][light:059]:   Red: 100%, Green: 89%, Blue: 71%
[12:46:48][D][micro_wake_word:399]: Resetting buffers and probabilities
[12:46:48][D][micro_wake_word:195]: State changed from IDLE to START_MICROPHONE
[12:46:48][W][component:237]: Component script took a long time for an operation (74 ms).
[12:46:48][W][component:238]: Components should block for at most 30 ms.
[12:46:48][D][micro_wake_word:107]: Starting Microphone
[12:46:48][D][micro_wake_word:195]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[12:46:48][D][esp-idf:000]: I (576293) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[12:46:48][D][micro_wake_word:195]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
```
I can download the wav from browser and it plays fine.
http://{local_ip}/api/esphome/ffmpeg_proxy/cb46e1fcd1ebaf169af02d29a19fe897/RtyAXHJ6e7cMuNiomlNi0w.wav

darinlarimore, 2025-02-27:
> I tried to use [m5stack-atom-echo.yaml](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml?rgh-link-date=2025-02-27T18%3A30%3A00.000Z) and it crashed HA, not kidding.

Yeah, It took over 30 min for mine to initially build on a Raspberry pi 4 using that Yaml! But it finally worked and subsequent builds are much faster.

hitnrun30, 2025-02-27:
> Yeah, It took over 30 min for mine to initially build on a Raspberry pi 4 using that Yaml! But it finally worked and subsequent builds are much faster.

I get ya, but I mean the machine shutdown.

nem0nic, 2025-03-21:
I'm having the same issue as @Kalekulan above

```
[10:48:38][D][esp-idf:000][ann_read]: E (50050) esp-tls: [sock=61] select() timeout
[10:48:38][D][esp-idf:000][ann_read]: E (50059) TRANSPORT_BASE: Failed to open a new connection: 32774
[10:48:38][D][esp-idf:000][ann_read]: E (50064) HTTP_CLIENT: Connection failed, sock < 0
[10:48:38][E][speaker_media_player.pipeline:112]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
```

Kalekulan, 2025-04-11:
@nem0nic  I managed to solve my problem. 

Going to your homeassistant instance (if you have one):
Settings -> Network. 

Change the local IP address to the host IP address of the server you are running it on. Mine was set to the Docker network IP address:

[image]